bootlin/buildroot-external-st #22: U-Boot compilation fails with Error: You must add new CONFIG options using Kconfig

FrankBau, 2022-05-30:
Thank you for providing this repo.

On a Linux 5.4.0-113-generic #127~18.04.1-Ubuntu, I do in an empty folder:
```
git clone -b st/2021.02 https://github.com/bootlin/buildroot.git
git clone -b st/2021.02 https://github.com/bootlin/buildroot-external-st.git
cd buildroot
make BR2_EXTERNAL=../buildroot-external-st st_stm32mp157c_dk2_defconfig
make
```
which fails with
```
>>> uboot v2020.10-stm32mp-r2 Building
...
  HOSTCC  scripts/dtc/dtc-parser.tab.o
  HOSTCC  scripts/dtc/dtc-lexer.lex.o
  LDS     u-boot.lds
Error: You must add new CONFIG options using Kconfig
The following new ad-hoc CONFIG options were detected:
CONFIG_FS_FAT

Please add these via Kconfig instead. Find a suitable Kconfig
file and add a 'config' or 'menuconfig' option.
Makefile:1037: recipe for target '.binman_stamp' failed
```

The same happened for `st_stm32mp157c_dk2_demo_defconfig`

Regards

Frank

kmaincent, 2022-05-30:
Hello, weird, I just tried on our build server which is a Ubuntu 20.04 and it built it well.
It seems to be a config error, but I don't know why.
Are you sure you don't change anything?

FrankBau, 2022-05-30:
Well, I found a debian server where uboot builds fine and try to find the diffs. Weird indeed.
Frank

FrankBau, 2022-05-31:
I retried, but the error persists. Tried to compare to the successfull build on that Debian server, but gave up. Could attach verbose build logs if that helps
Regards Frank

kmaincent, 2022-06-01:
Yes please do so, maybe I can find something relevant.

tpetazzoni, 2022-07-08:
Could you give more details about the version of the distribution with which it is failing? Are you able to reproduce in a clean Docker container for that distribution?

FrankBau, 2022-07-10:
```
$ cat /etc/os-release 
NAME="Ubuntu"
VERSION="18.04.6 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.6 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
```

it does not fail on a clean `docker run -i -t ubuntu:18.04 /bin/bash` container. So my failing Ubuntu 18.04 system is somehow "special".

tpetazzoni, 2022-07-13:
It's going to be difficult to help without being able to reproduce. Are you perhaps able to narrow down what is the difference between the "clean" Docker container and your environment?

FrankBau, 2022-07-14:
I'm fine with using another machine and you can close the issue as you like. Still don't understand what step produces the "Error: You must add new CONFIG options using Kconfig" or how I could reproduce that error from the command line.

closed 2022-12-13 by kmaincent