bootlin/buildroot-external-st #23: Linux kernel compilation fails with scripts/extract-cert.c:21:10: fatal error: openssl/bio.h: No such file or directory

FrankBau, 2022-05-30:
Can be easily fixed by adding `libssl-dev` to the pre-requisites in https://github.com/bootlin/buildroot-external-st#pre-requisites.
Found on a Linux 4.19.0-18-amd64 #1 SMP Debian 4.19.208-1 server as build host.
hth
KnarfB

tpetazzoni, 2022-05-30:
Not the right fix. Buildroot can build host-openssl as a dependency of Linux by enabling BR2_LINUX_KERNEL_NEEDS_HOST_OPENSSL=y.

kmaincent, 2022-05-31:
Issue already reported here: #17 
Are you using the same kernel Configuration as the one in the BR2_EXTERNAL?

FrankBau, 2022-05-31:
Yes, I did a clean build in a new folder:
```
git clone -b st/2021.02 https://github.com/bootlin/buildroot.git
git clone -b st/2021.02 https://github.com/bootlin/buildroot-external-st.git
cd buildroot
make BR2_EXTERNAL=../buildroot-external-st st_stm32mp157c_dk2_defconfig
make
```
on that Debian server I use as workaround for #22.

tpetazzoni, 2022-05-31:
_build tainted by host openssl_ => this is inherent to cross-compilation in general, it's impossible to fully isolate from what's on the host machine
_BR2_LINUX_KERNEL_NEEDS_HOST_OPENSSL missing in the defconfigs_ => yes, I agree

kmaincent, 2022-05-31:
It is indeed an issue, I just add the config to fix it, thanks.

closed 2022-05-31 by kmaincent