bootlin/buildroot-external-st #29: Failed to compile the sdcard.img

VectorForceResearch, 2022-09-25:
Hi Bootlin, 
I'm following following steps for my STM32MP157F-DK2. I want to use the GPIO for the R.Pi and Arduino connectors.

I went through the original steps without issue here: https://bootlin.com/blog/building-a-linux-system-for-the-stm32mp1-basic-system/
In the output/images, it appears to have all the files referenced including the sdcard.img. 

Whilst it was running through the long build process, I discovered the updated method here: https://bootlin.com/blog/updated-buildroot-support-for-stm32mp1-platforms/

When the original process was completed, I renamed the buildroot folder as buildroot_org. Then started the updated process steps.
It did not complete as it had build errors and says to post them.

/home/jim/buildroot/output/build/m4projects-1.5.0/Projects/STM32MP157C-DK2/Demonstrations/AI_Character_Recognition/Src/main.c:209:3: internal compiler error: Aborted
  209 |   log_info("start ai nn processing\n");
      |   ^~~~~~~~
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://bugs.linaro.org/> for instructions.
make[2]: *** [/home/jim/buildroot/output/build/m4projects-1.5.0/Makefile.stm32:64: /home/jim/buildroot/output/build/m4projects-1.5.0/STM32MP157C-DK2/Demonstrations/AI_Character_Recognition/out/Debug//home/jim/buildroot/output/build/m4projects-1.5.0/Projects/STM32MP157C-DK2/Demonstrations/AI_Character_Recognition/Src/main.c.o] Error 1
make[1]: *** [package/pkg-generic.mk:292: /home/jim/buildroot/output/build/m4projects-1.5.0/.stamp_built] Error 2
make: *** [Makefile:84: _all] Error 2
[jim@octopus buildroot]$

I missed something. Not sure if I was supposed to duplicate some of the steps for the menuconfig? or maybe one of the make steps needs to be duplicated? 

Thank you,
Jim

kmaincent, 2022-09-26:
Hello, what configuration are you using?
`I missed something. Not sure if I was supposed to duplicate some of the steps for the menuconfig? or maybe one of the make steps needs to be duplicated?`
What do you mean by that? Are you trying to update your old defconfig?

About the error, as you can see it comes from the build of m4projects package which builds examples for the m4 CPU core. Have you some log before that which give more information about the errors?

tpetazzoni, 2022-09-26:
@kmaincent please backport https://gitlab.com/buildroot.org/buildroot/-/commit/f1017a5834b744e6f87c3372a6de60d553566fd2 into our Buildroot branch. This should fix this issue.

kmaincent, 2022-09-26:
Ok thanks for the information, the rollback has been done.
@VectorForceResearch please pull the [Buildroot repo](https://github.com/bootlin/buildroot) and try again.

closed 2022-09-27 by kmaincent