bootlin/buildroot-external-st #30: The OpenAMP_TTY_echo test did not run correctly

aronfeher, 2022-12-26:
Hi Bootlin!

I've got my hands on an STM32MP157D-DK1 board, and I wanted to start the development of an OpenAMP application.
I've built the image according to the readme using the _st_stm32mp157d_dk1_demo_defconfig_. So far so good. I managed to boot up, and to connect to the board using SSH, the GPIO_EXTI OpenAMP_raw examples work as intended.

When I start the OpenAMP_TTY_echo, the following lines are in dmesg:
```
[  136.626935] remoteproc remoteproc0: powering up m4
[  136.631240] remoteproc remoteproc0: Booting fw image OpenAMP_TTY_echo.elf, size 197920
[  136.638985]  remoteproc0#vdev0buffer: assigned reserved memory node vdev0buffer@10042000
*[  136.653920] virtio_rpmsg_bus virtio0: creating channel rpmsg-tty-channel addr 0x400
[  136.660432] virtio_rpmsg_bus virtio0: rpmsg host is online
*[  136.663712] virtio_rpmsg_bus virtio0: creating channel rpmsg-tty-channel addr 0x401
[  136.665998]  remoteproc0#vdev0buffer: registered virtio0 (type 7)
[  136.679447] remoteproc remoteproc0: remote processor m4 is now up
```

Here at the * lines, the ttyRPMSG0/1 is not shown as it should be according to the examples.
Further more, there is no _/dev/ttyRPMSG0_ or _/dev/ttyRPMSG1_ present,  and the _/sys/kernel/debug/remoteproc/remoteproc0/trace0_ is also missing.

Did I miss something during the default configuration, or there is some hidden instruction that I did not find to make the M4-A7 communication work?

Thank you in advance,
Aron.

tpetazzoni, 2022-12-27:
Thanks for your report. My colleague @kmaincent will have a look next week.

kmaincent, 2023-01-03:
Indeed it does not work well. I face the same issue here.
The  `/sys/kernel/debug/remoteproc/remoteproc0/trace0` is present when loading the firmware but no `/dev/ttyRPMSG*`

kmaincent, 2023-01-03:
The m4 projects package was not updated to the last version.
I did it and pushed it: 1acf3c7c53da09abdaf26c671074ff2f37782f9e

closed 2023-01-03 by kmaincent

aronfeher, 2023-01-03:
Thank you very much for this quick fix. As soon as I'm out of work I'll try it out.