bootlin/buildroot-external-st #33: STM32MP157D-DK1 Demo does not boot

hreysenbach, 2023-03-05:
When using the `st_stm32mp157d_dk1_demo_defconfig` the STM32MP157D-DK1 does not boot. 

ARM Trusted Firmware starts up but stops with the following logs:
```
NOTICE:  CPU: STM32MP157DAC Rev.Z
NOTICE:  Model: STMicroelectronics STM32MP157D-DK1 Discovery Board
NOTICE:  Board: MB1272 Var3.0 Rev.C-02
NOTICE:  BL2: v2.6-stm32mp1-r2.0(release):2021.05-5963-ga40218f3c9(a40218f3)
NOTICE:  BL2: Built : 22:34:19, Mar  5 2023
NOTICE:  BL2: Booting BL32
```

When I try to build it in debug mode (in the bootloaders menu in menuconfig) it fails to compile because the `xlat_table will not fit in region RAM` and that it overflowed by 4096 bytes.

I'm not too sure where to go from here. I've tried to modify ARM-Trusted Firmware to get more verbose logging (mainly through the Makefile) and I can get more but nothing that is really useful. I assume the issue is more on the OP-TEE side but that doesn't seem to have any debug logs... Any help would be appreciated

hreysenbach, 2023-03-06:
So I double-checked my branches just to make sure and they were both on the `2022.02.7` branch. I did a clean build and it started working so that's good news. Thanks for the quick response and help. Just something to note, with the clean build it had the same commit for ARM Trusted Firmware (`NOTICE:  BL2: v2.6-stm32mp1-r2.0(release):2021.05-5963-ga40218f3c9(a40218f3)`). 

I'm also trying to bring up a custom board (based on the STM32MP157D-DK1) and so I am still curious if you can build ARM Trusted Firmware in debug mode?

kmaincent, 2023-03-09:
Mmh weird that the version is still 2021.05 I think it should match the last Buildroot tag.

About the DEBUG mode, you might be able to enable it by disabling other TFA features. Only a few features can be enabled together to fit in the region RAM.

hreysenbach, 2023-03-10:
Unfortunately you'd probably know more than me on the version number.

Thanks for the help and the guidance :smile:

closed 2023-03-10 by hreysenbach

frenchie68, 2023-03-13:
I do not think this issue should be closed. The version confusion persists. Shortly after:

>>> arm-trusted-firmware custom Installing to images directory

I get the following diagnostics:

	echo "NAME=Buildroot"; \
	echo "VERSION=2021.05-5963-ga40218f3c9"; \
	echo "ID=buildroot"; \
	echo "VERSION_ID=2022.02.7"; \
	echo "PRETTY_NAME=\"Buildroot 2022.02.7\"" \
[...]
>>>   Sanitizing RPATH in target tree
PER_PACKAGE_DIR=/home/mit/EmbeddedLinux/bootlin/systemd-extSDK/per-package /home/mit/EmbeddedLinux/bootlin/buildroot/support/scripts/fix-rpath target
>>>   Sanity check in overlay /home/mit/EmbeddedLinux/bootlin/buildroot-external-st/board/stmicroelectronics/stm32mp1/overlay/
ERROR: The overlay in /home/mit/EmbeddedLinux/bootlin/buildroot-external-st/board/stmicroelectronics/stm32mp1/overlay/ is not using a merged /usr for the following directories: /lib
make[1]: *** [Makefile:755: target-finalize] Error 1
make: *** [Makefile:23: _all] Error 2

So, besides the version confusion, something seems to be wrong with the contents of board/stmicroelectronics/stm32mp1/overlay when BR2_ROOTFS_MERGED_USR is 'y'

Additional information:
mit@m211mp157:~/EmbeddedLinux/bootlin/buildroot$ git status -b
On branch st/2022.02.7
Your branch is up to date with 'origin/st/2022.02.7'.
mit@m211mp157:~/EmbeddedLinux/bootlin/buildroot-external-st$ git status -b
On branch st/2022.02.7
Your branch is up to date with 'origin/st/2022.02.7'.