bootlin/buildroot-external-st #34: Enable GPIO on STM32MP157F-DK2

AaronMill93, 2023-03-12:
Hi,

I'm following the STM32CubeMX devicetree doc in an attempt to enable SPI for the STM32MP157F-DK2.
I was able to figure out the SPI pins and added the necessary code as per tutorial. I'm just wondering if there's anything extra needed to enable a GPIO pin for SS (e.g. PA8)? I did a diff of the generated code from CubeMX with and without activating PA8 but there's seems to be no changes. Any help is appreciated.

Aaron

kmaincent, 2023-05-17:
Hello, 
In the devicetree you will need to specify the gpio in the spi-bus node with the cs-gpio parameter:
https://elixir.bootlin.com/linux/v5.15.111/source/Documentation/devicetree/bindings/spi/spi-controller.yaml#L28
And configure the GPIO pin for SS in the pinctrl with ANALOG function, here are few examples:
https://elixir.bootlin.com/linux/v5.15.111/source/arch/arm/boot/dts/stm32mp15-pinctrl.dtsi

AaronMill93, 2023-05-28:
I figured this out myself, but thank you for the reply. From my experience, there is currently no way to use SPI functionality on the current branch. /dev/spi0.0 will only be populated after the instructions from ST are followed properly and buildroot from branch st/2021.02 is used.

kmaincent, 2023-08-29:
Done dc13f5a

closed 2023-08-29 by kmaincent