bootlin/buildroot-external-st #40: Error in buildroot-external-st (RTC, WLAN)

MarioPieschel, 2023-08-18:
Error in buildroot-external-st, branch 2023.02.2. The result is that the WLAN will not work.
In branch 2022.02.7 the RTC is initialized correctly, and WLAN works.

My board is the STM32MP135F-DK, MB1635B. With this configuration: st_stm32mp135f_dk_demo_defconfig.

dmesg | grep rtc
[    3.084346] stm32_rtc 5c004000.rtc: rtc_ck is slow
[    3.187767] stm32_rtc 5c004000.rtc: Can't enter in init mode. Prescaler config failed.
[    3.194423] stm32_rtc: probe of 5c004000.rtc failed with error -110

brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x5

[kernel_log.txt](https://github.com/bootlin/buildroot-external-st/files/12378110/kernel_log.txt)

kmaincent, 2023-11-03:
Hello, thanks for reporting the issue.
Indeed there is an issue with the devicetree TF-A generated by Cube MX.
The devicetree node "rcc" is wrongly set.
I am reporting the issue to ST and will fix it soon.
meanwhile, you can import the [devicetree](https://github.com/STMicroelectronics/arm-trusted-firmware/blob/v2.8-stm32mp/fdts/stm32mp135f-dk.dts) in the Buildroot external instead of the one generated by CubeMX.

kmaincent, 2023-11-08:
ST raised a ticket on this issue.
I will add the devicetree fix in the next release.

MarioPieschel, 2023-11-10:
FYI

Mit freundlichen Grüßen / Best regards 
Mario Pieschel 
embeddeers GmbH
Am Borsigturm 70
13507 Berlin
tel: +49 30 2000 580-31
fax: +49 30 2000 580-99
mobil: +49 XXX XXX XX XX
Geschäftsführer: Christian Zieme, Mandy Andritzke
Handelsregister: Amtsgericht Berlin-Charlottenburg, HRB 107372 B 
***@***.***
 https://www.embeddeers.com  
 	   	   	   	   	   
 
 Informationen zum Datenschutz nach Art. 13 und Art. 14 DSGVO finden Sie unter: https://www.embeddeers.com/datenschutzerklaerung/