bootlin/buildroot-external-st #41: default st_stm32mp157c_dk2_defconfig fails with

mstaack, 2023-08-18:
compilation error:

```
>>> arm-trusted-firmware custom Building
if test -d /home/max/buildroot/output/build/arm-trusted-firmware-custom/tools/fiptool; then PATH="/home/max/buildroot/output/host/bin:/home/max/buildroot/output/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" AR="/home/max/buildroot/output/host/bin/arm-none-linux-gnueabihf-gcc-ar" AS="/home/max/buildroot/output/host/bin/arm-none-linux-gnueabihf-as" LD="/home/max/buildroot/output/host/bin/arm-none-linux-gnueabihf-ld" NM="/home/max/buildroot/output/host/bin/arm-none-linux-gnueabihf-gcc-nm" CC="/home/max/buildroot/output/host/bin/arm-none-linux-gnueabihf-gcc" GCC="/home/max/buildroot/output/host/bin/arm-none-linux-gnueabihf-gcc" CPP="/home/max/buildroot/output/host/bin/arm-none-linux-gnueabihf-cpp" CXX="/home/max/buildroot/output/host/bin/arm-none-linux-gnueabihf-g++" FC="/home/max/buildroot/output/host/bin/arm-none-linux-gnueabihf-gfortran" F77="/home/max/buildroot/output/host/bin/arm-none-linux-gnueabihf-gfortran" RANLIB="/home/max/buildroot/output/host/bin/arm-none-linux-gnueabihf-gcc-ranlib" READELF="/home/max/buildroot/output/host/bin/arm-none-linux-gnueabihf-readelf" STRIP="/home/max/buildroot/output/host/bin/arm-none-linux-gnueabihf-strip" OBJCOPY="/home/max/buildroot/output/host/bin/arm-none-linux-gnueabihf-objcopy" OBJDUMP="/home/max/buildroot/output/host/bin/arm-none-linux-gnueabihf-objdump" AR_FOR_BUILD="/usr/bin/ar" AS_FOR_BUILD="/usr/bin/as" CC_FOR_BUILD="/usr/bin/gcc" GCC_FOR_BUILD="/usr/bin/gcc" CXX_FOR_BUILD="/usr/bin/g++" LD_FOR_BUILD="/usr/bin/ld" CPPFLAGS_FOR_BUILD="-I/home/max/buildroot/output/host/include" CFLAGS_FOR_BUILD="-O2 -I/home/max/buildroot/output/host/include" CXXFLAGS_FOR_BUILD="-O2 -I/home/max/buildroot/output/host/include" LDFLAGS_FOR_BUILD="-L/home/max/buildroot/output/host/lib -Wl,-rpath,/home/max/buildroot/output/host/lib" FCFLAGS_FOR_BUILD="" DEFAULT_ASSEMBLER="/home/max/buildroot/output/host/bin/arm-none-linux-gnueabihf-as" DEFAULT_LINKER="/home/max/buildroot/output/host/bin/arm-none-linux-gnueabihf-ld" CPPFLAGS="-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64" CFLAGS="-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -Os -g0 -D_FORTIFY_SOURCE=1" CXXFLAGS="-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -Os -g0 -D_FORTIFY_SOURCE=1" LDFLAGS="" FCFLAGS=" -Os -g0" FFLAGS=" -Os -g0" PKG_CONFIG="/home/max/buildroot/output/host/bin/pkg-config" STAGING_DIR="/home/max/buildroot/output/host/arm-buildroot-linux-gnueabihf/sysroot" INTLTOOL_PERL=/usr/bin/perl /usr/bin/make -j6 -C /home/max/buildroot/output/build/arm-trusted-firmware-custom/tools/fiptool CROSS_COMPILE="/home/max/buildroot/output/host/bin/arm-none-linux-gnueabihf-" STM32MP_SDMMC=1 AARCH32_SP=optee DTB_FILE_NAME=stm32mp157c-dk2.dtb BL33_CFG=/home/max/buildroot/output/images/u-boot.dtb STM32MP_USB_PROGRAMMER=1 STM32MP1_OPTEE_IN_SYSRAM=1 PLAT=stm32mp1 TARGET_BOARD= ARM_ARCH_MAJOR=7 ARCH=aarch32 BL32=/home/max/buildroot/output/images/tee-header_v2.bin BL32_EXTRA1=/home/max/buildroot/output/images/tee-pager_v2.bin BL32_EXTRA2=/home/max/buildroot/output/images/tee-pageable_v2.bin AARCH32_SP=optee BL33=/home/max/buildroot/output/images/u-boot-nodtb.bin CPPFLAGS="-I/home/max/buildroot/output/host/include" LDLIBS="-L/home/max/buildroot/output/host/lib -Wl,-rpath,/home/max/buildroot/output/host/lib -lcrypto" ; fi
/usr/bin/gcc -c -I/home/max/buildroot/output/host/include -D_GNU_SOURCE -D_XOPEN_SOURCE=700 -Wall -Werror -pedantic -std=c99 -O2 -DUSING_OPENSSL3=1 -DPLAT_DEF_FIP_UUID -I../../include/tools_share  -I/usr/include -I../../plat/st/stm32mp1//include -I./ ../../plat/st/stm32mp1/plat_def_uuid_config.c -o plat_def_uuid_config.o
PATH="/home/max/buildroot/output/host/bin:/home/max/buildroot/output/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" CFLAGS="-fno-stack-protector -fno-PIE" /usr/bin/make -j6 -C /home/max/buildroot/output/build/arm-trusted-firmware-custom CROSS_COMPILE="/home/max/buildroot/output/host/bin/arm-none-linux-gnueabihf-" STM32MP_SDMMC=1 AARCH32_SP=optee DTB_FILE_NAME=stm32mp157c-dk2.dtb BL33_CFG=/home/max/buildroot/output/images/u-boot.dtb STM32MP_USB_PROGRAMMER=1 STM32MP1_OPTEE_IN_SYSRAM=1 PLAT=stm32mp1 TARGET_BOARD= ARM_ARCH_MAJOR=7 ARCH=aarch32 BL32=/home/max/buildroot/output/images/tee-header_v2.bin BL32_EXTRA1=/home/max/buildroot/output/images/tee-pager_v2.bin BL32_EXTRA2=/home/max/buildroot/output/images/tee-pageable_v2.bin AARCH32_SP=optee BL33=/home/max/buildroot/output/images/u-boot-nodtb.bin all fip
Including bl32/optee/optee.mk
Trusted Firmware-A built for OP-TEE payload support
  HOSTCC  stm32image.c
Building stm32mp1
  CC      lib/libfdt/fdt.c
  CC      lib/libfdt/fdt_addresses.c
  CC      lib/libfdt/fdt_empty_tree.c
  CC      lib/libfdt/fdt_ro.c
  CC      lib/libfdt/fdt_rw.c
  HOSTCC  fiptool.c
  CC      lib/libfdt/fdt_strerror.c
In file included from fiptool.h:16,
                 from fiptool.c:19:
fiptool_platform.h:19:11: fatal error: openssl/sha.h: No such file or directory
   19 | # include <openssl/sha.h>
      |           ^~~~~~~~~~~~~~~
compilation terminated.
  CC      lib/libfdt/fdt_sw.c
make[3]: *** [Makefile:76: fiptool.o] Error 1
make[2]: *** [Makefile:1562: tools/fiptool/fiptool] Error 2
make[2]: *** Waiting for unfinished jobs....
  HOSTLD  stm32image

Built stm32image successfully

make[1]: *** [package/pkg-generic.mk:293: /home/max/buildroot/output/build/arm-trusted-firmware-custom/.stamp_built] Error 2
make: *** [Makefile:82: _all] Error 2
```

setup as in readme:

```
max@max:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.3 LTS
Release:	22.04
Codename:	jammy

max@max:~$ uname -a
Linux max 5.15.0-79-generic #86-Ubuntu SMP Mon Jul 10 16:11:29 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux

max@max:~$ sudo apt install debianutils sed make binutils build-essential gcc g++ bash patch gzip bzip2 perl tar cpio unzip rsync file bc git
[sudo] password for max:
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
bash is already the newest version (5.1-6ubuntu1).
bc is already the newest version (1.07.1-3build1).
build-essential is already the newest version (12.9ubuntu3).
bzip2 is already the newest version (1.0.8-5build1).
cpio is already the newest version (2.13+dfsg-7).
debianutils is already the newest version (5.5-1ubuntu2).
file is already the newest version (1:5.41-3).
g++ is already the newest version (4:11.2.0-1ubuntu1).
gcc is already the newest version (4:11.2.0-1ubuntu1).
make is already the newest version (4.3-4.1build1).
patch is already the newest version (2.7.6-7build2).
sed is already the newest version (4.8-1ubuntu2).
binutils is already the newest version (2.38-4ubuntu2.3).
git is already the newest version (1:2.34.1-1ubuntu1.10).
gzip is already the newest version (1.10-4ubuntu4.1).
perl is already the newest version (5.34.0-3ubuntu1.2).
[... 12 lines cut ...]
Resolving deltas: 100% (338240/338240), done.

max@max:~$ git clone -b st/2023.02.2 https://github.com/bootlin/buildroot-external-st.git
Cloning into 'buildroot-external-st'...
remote: Enumerating objects: 929, done.
remote: Counting objects: 100% (393/393), done.
remote: Compressing objects: 100% (78/78), done.
remote: Total 929 (delta 315), reused 352 (delta 300), pack-reused 536
Receiving objects: 100% (929/929), 215.68 KiB | 3.48 MiB/s, done.
Resolving deltas: 100% (580/580), done.

max@max:~$ cd buildroot/
max@max:~/buildroot$ make BR2_EXTERNAL=../buildroot-external-st st_stm32mp157c_dk2_defconfig
mkdir -p /home/max/buildroot/output/build/buildroot-config/lxdialog
PKG_CONFIG_PATH="" make CC="/usr/bin/gcc" HOSTCC="/usr/bin/gcc" \
    obj=/home/max/buildroot/output/build/buildroot-config -C support/kconfig -f Makefile.br conf
/usr/bin/gcc -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE  -I/home/max/buildroot/output/build/buildroot-config -DCONFIG_=\"\"  -MM *.c > /home/max/buildroot/output/build/buildroot-config/.depend 2>/dev/null || :
/usr/bin/gcc -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE  -I/home/max/buildroot/output/build/buildroot-config -DCONFIG_=\"\"   -c conf.c -o /home/max/buildroot/output/build/buildroot-config/conf.o
[... cut ...] d/buildroot-config/zconf.tab.o
/usr/bin/gcc -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE  -I/home/max/buildroot/output/build/buildroot-config -DCONFIG_=\"\"   /home/max/buildroot/output/build/buildroot-config/conf.o /home/max/buildroot/output/build/buildroot-config/zconf.tab.o  -o /home/max/buildroot/output/build/buildroot-config/conf
rm /home/max/buildroot/output/build/buildroot-config/zconf.tab.c
#
# configuration written to /home/max/buildroot/.config
#
```

mstaack, 2023-08-18:
solved by addying missing openssl lib with:
```
sudo apt install libssl-dev
```

tpetazzoni, 2023-08-18:
I'm reopening, because installing libssl-dev on the host is not the proper solution, Buildroot is supposed to build its own host libopenssl. I will address that.

mstaack, 2023-08-28:
@tpetazzoni is this solved?

kmaincent, 2023-08-29:
Last commit dfb89e5 should fix it.
Thanks for the bug report.

closed 2023-09-05 by mstaack