bootlin/buildroot-external-st #43: SPI5 Failed to create device link (0x180)

MarioPieschel, 2023-08-25:
I have configured SPI5 in Stm32CubeMX
According to buildroot-external-st/docs/stm32cubemx.md
I copied the generated code in the folders
 kernel -> kernel-dts
 u-boot -> uboot-dts
 tf-a -> tfa-dts
 optee-os -> optee-dts

Unfortunately the description is missing, how to compile this.

by guessing:
make uboot-rebuild && make optee-rebuild && make linux-rebuild all

Is it correct?

kernel message:
[    0.059031] stm32-sys-bus 5c007000.etzpc: Failed to create device link (0x180) with soc:pinctrl@50002000
[    0.445547] stm32-sys-bus 5c007000.etzpc: Failed to create device link (0x180) with 50000000.rcc
[    0.468479] stm32-sys-bus 5c007000.etzpc: Failed to create device link (0x180) with soc:pinctrl@50002000

I guess, the clock for SPI5 is not enabled.

My board is the STM32MP135F-DK, MB1635B. With this configuration: st_stm32mp135f_dk_demo_defconfig.

rename it to dk.ico
[dk.ioc.txt](https://github.com/bootlin/buildroot-external-st/files/12437529/dk.ioc.txt)
[kernel_log_SPI5_Failed.txt](https://github.com/bootlin/buildroot-external-st/files/12437536/kernel_log_SPI5_Failed.txt)

MarioPieschel, 2023-08-25:
make uboot-rebuild && make optee-os-rebuild && make linux-rebuild all

MarioPieschel, 2023-08-29:
my generated files:
[dk.zip](https://github.com/bootlin/buildroot-external-st/files/12461425/dk.zip)

MarioPieschel, 2023-08-31:
If I try SPI5v configure with openSTLinux I will get a **kernel panic**.
I gess there is the optee-os source code buggy.

Starting kernel ...

E/TC:0   Panic 'Watchdog' at /usr/src/debug/optee-os-stm32mp/3.19.0-stm32mp-r1-r0/core/drivers/stm32_iwdg.c:198 <stm32_iwdg_it_handler>
E/TC:0   TEE load address @ 0xde000000
E/TC:0   Call stack:
E/TC:0    0xde0030f9
E/TC:0    0xde019f87
E/TC:0    0xde00c873
E/TC:0    0xde01966d
E/TC:0    0xde039601
E/TC:0    0xde000340

kmaincent, 2023-08-31:
> make uboot-rebuild && make optee-os-rebuild && make linux-rebuild all

And  `make arm-trusted-firmware-rebuild` but yes it is correct.
For your SPI issue, I will point this issue to ST. They will be more experienced to help you with that.

cpriouzeau, 2023-08-31:
For recompiling the component: 
    make uboot-rebuild && make optee-rebuild && make optee-rebuild && make arm-trusted-firmware-rebuild && make linux-rebuild all

Then flash: 
 - tf-a-stm32mp135f-dk.stm32 on fsbl1 partition
 - fip.bin on fip partition
 - put the new kernel devicetree on rootfs partition (boot directory)

cpriouzeau, 2023-08-31:
We have detected several issue: 
- on optee-os definition of SPI5 (must be on NS non-secure)
- on U-BOOT and on Kernel, the pinctrl have a bad alternate
- on kernel, the compatible for spidev is not correct.

Optee: 
- You must have only the declaration of SPI5 on NS
@@ -269,6 +269,7 @@
 	DECPROT(STM32MP1_ETZPC_LTDC_ID, DECPROT_NS_RW, DECPROT_UNLOCK)
 	DECPROT(STM32MP1_ETZPC_SDMMC1_ID, DECPROT_NS_RW, DECPROT_UNLOCK)
 	DECPROT(STM32MP1_ETZPC_SDMMC2_ID, DECPROT_NS_RW, DECPROT_UNLOCK)
+	DECPROT(STM32MP1_ETZPC_SPI5_ID, DECPROT_NS_RW, DECPROT_UNLOCK)
 	DECPROT(STM32MP1_ETZPC_USART2_ID, DECPROT_NS_RW, DECPROT_UNLOCK)
 	DECPROT(STM32MP1_ETZPC_OTG_ID, DECPROT_NS_RW, DECPROT_UNLOCK)
 	DECPROT(STM32MP1_ETZPC_USBPHYCTRL_ID, DECPROT_NS_RW, DECPROT_UNLOCK)

U-BOOT: 
Use the correct pinctrl: 
+	spi5_pins_mx: spi5_mx-0 {
+		pins1 {
+			pinmux = <STM32_PINMUX('H', 7, AF6)>, /* SPI5_SCK */
+				 <STM32_PINMUX('H', 3, AF5)>; /* SPI5_MOSI */
+			bias-disable;
+			drive-push-pull;
+			slew-rate = <1>;
+		};
+
+		pins2 {
+			pinmux = <STM32_PINMUX('A', 8, AF5)>; /* SPI5_MISO */
+			bias-disable;
+		};
+	};
+
+	spi5_sleep_pins_mx: spi5_sleep_mx-0 {
+		pins {
+			pinmux = <STM32_PINMUX('A', 8, ANALOG)>, /* SPI5_MISO */
+					 <STM32_PINMUX('H', 3, ANALOG)>, /* SPI5_MOSI */
+					 <STM32_PINMUX('H', 7, ANALOG)>; /* SPI5_SCK */
+		};
+	};
+&spi5{
+	pinctrl-names = "default", "sleep";
+	pinctrl-0 = <&spi5_pins_mx>;
+	pinctrl-1 = <&spi5_sleep_pins_mx>;
+	status = "okay";
+
+	/* USER CODE BEGIN spi5 */
+	/* USER CODE END spi5 */
+};

On Kernel: 
Use the correct pinctrl + add a valid compatible for spidev
+	spi5_pins_mx: spi5_mx-0 {
+		pins1 {
+			pinmux = <STM32_PINMUX('H', 7, AF6)>, /* SPI5_SCK */
+				 <STM32_PINMUX('H', 3, AF5)>; /* SPI5_MOSI */
+			bias-disable;
+			drive-push-pull;
+			slew-rate = <1>;
+		};
+		pins2 {
+			pinmux = <STM32_PINMUX('A', 8, AF5)>; /* SPI5_MISO */
+			bias-disable;
+		};
+
+	};
+
+	spi5_sleep_pins_mx: spi5_sleep_mx-0 {
+		pins {
+			pinmux = <STM32_PINMUX('A', 8, ANALOG)>, /* SPI5_MISO */
+					 <STM32_PINMUX('H', 3, ANALOG)>, /* SPI5_MOSI */
+					 <STM32_PINMUX('H', 7, ANALOG)>; /* SPI5_SCK */
+		};
+	};
+
+&spi5{
+	pinctrl-names = "default", "sleep";
+	pinctrl-0 = <&spi5_pins_mx>;
+	pinctrl-1 = <&spi5_sleep_pins_mx>;
+	cs-gpios = <&gpioh 11 0>; /* PH11 */
+	status = "okay";
+	spidev@0 {
+		status = "okay";
+		compatible = "linux,spidev";
+		reg = <0>;
+		spi-max-frequency = <30000000>;
+	};
+
+	/* USER CODE BEGIN spi5 */
+	/* USER CODE END spi5 */
+};

One time, all the devicetree will be updated: 
make uboot-rebuild
make optee-os-rebuild
make arm-trusted-firmware-rebuild (mainly to re-generate the fip.bin)
make linux-rebuild all

Then flash:
- tf-a-stm32mp135f-dk.stm32 on fsbl1 partition
- fip.bin on fip partition
- put the new kernel devicetree on rootfs partition (boot directory)

MarioPieschel, 2023-09-01:
Thank you very much for the very good help.

But the I have installed the STM32MP1 Distribution Package.
According to these instructions:
https://wiki.st.com/stm32mpu/wiki/STM32MP1_Distribution_Package#Building_the_OpenSTLinux_distribution

There is described the way with the bitbake command.

I don't know how to apply the make commands.

>make linux-rebuild all
make: *** No rule for make target 'linux-rebuild'.  Stop.
mpieschel@linux-lab-computer:~/st

Do I need to set an environment variable? In which folder(s) do the make commands have to be executed?

I am new to OpenSTLinux.

Thanks a lot!

kmaincent, 2023-09-01:
I thinks your are mixing the build systems.
The OpenSTLinux use Yocto build system where in this github we use Buildroot build system.
bitbake is a Yocto command, but `make linux-rebuild all` is for Buildroot.
You have to choose which build system you want to use, you can't use both together.
cpriouzeau answered you for the Buildroot Build System.

MarioPieschel, 2023-09-01:
Yes, sorry.

MarioPieschel, 2023-09-01:
Many thanks to cpriouzeau and kmaincent
It woks :O)

# ls /dev/spi*
/dev/spidev0.0

closed 2023-09-04 by kmaincent