bootlin/buildroot-external-st #44: dd: failed to open 'output/images/sdcard.img': Operation not permitted (issue with sdcard.img)

jaygitdev, 2023-10-11:
I was trying to follow the instructions for a STM32MP157d-DK1 board but ran into an issue where I apparently can't do anything with the *.img file that was produced.  It's only 200MB or so (which seems small) and I can't get it to work with DD.  In fact, I can't even copy it to take it to another computer to try and burn with Belana Etcher or another tool like that.  

This is what I'm seeing when trying the last step:
```
sudo dd if=output/images/sdcard.img of=/dev/mmcblk0 bs=1M
dd: failed to open 'output/images/sdcard.img': Operation not permitted
```

These are the steps I followed to create said image:

```
	$ sudo apt install debianutils sed make binutils build-essential gcc g++ bash patch gzip bzip2 	perl tar cpio unzip rsync file bc git
	$ git clone -b st/2023.02.2 https://github.com/bootlin/buildroot.git
	$ git clone -b st/2023.02.2 https://github.com/bootlin/buildroot-external-st.git
	$ cd buildroot/
	$ make BR2_EXTERNAL=../buildroot-external-st st_stm32mp157d_dk1_demo_defconfig
	$ make
	$ sudo dd if=output/images/sdcard.img of=/mmcblk0 bs=1M
```

As I mentioned above, all signs seem to point to an issue with *.img (at least to me), though I'm not sure what I might've done wrong.  Maybe I missed something but I don't believe so.  Let me know if you need any additional info from me.  Thanks!

kmaincent, 2023-10-12:
This seems weird.
Did you verify the file permissions?
`ls -l output/images/sdcard.img`

jaygitdev, 2023-10-17:
Yeah, I tried a few different things last week without any luck, unfortunately.  I had tried using:
`chmod 777 output/images/sdcard.img `
But that didn't change anything, unfortunately.  I didn't see any errors . . . but there were what appeared to be thousands of lines that flashed in front of me so it's not impossible I missed it.  I saved the output of everything that was left in the terminal (it's not complete, but it's a 775 KB file).  Below is the last section, if it helps:
```
Superblock backups stored on blocks: 
	8193, 24577, 40961, 57345, 73729

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (4096 blocks): done
Copying files into the device: done
Writing superblocks and filesystem accounting information: done 

ln -sf rootfs.ext2 /home/local/a/b/Documents/Projects/STM32MP1/STBuildroot/buildroot/output/images/rootfs.ext4
ln -snf /home/local/a/b/Documents/Projects/STM32MP1/STBuildroot/buildroot/output/host/arm-buildroot-linux-gnueabihf/sysroot /home/local/a/b/Documents/Projects/STM32MP1/STBuildroot/buildroot/output/staging
>>>   Executing post-image script /home/local/a/b/Documents/Projects/STM32MP1/STBuildroot/buildroot-external-st/board/stmicroelectronics/stm32mp1/post-image.sh
INFO: cmd: "mkdir -p "/home/local/a/b/Documents/Projects/STM32MP1/STBuildroot/buildroot/output/build/genimage.tmp"" (stderr):
INFO: cmd: "rm -rf "/home/local/a/b/Documents/Projects/STM32MP1/STBuildroot/buildroot/output/build/genimage.tmp"/*" (stderr):
INFO: cmd: "mkdir -p "/home/local/a/b/Documents/Projects/STM32MP1/STBuildroot/buildroot/output/images"" (stderr):
INFO: hdimage(sdcard.img): adding partition 'fsbl1' (in MBR) from 'tf-a-stm32mp157d-dk1-mx.stm32' ...
INFO: hdimage(sdcard.img): adding partition 'fsbl2' (in MBR) from 'tf-a-stm32mp157d-dk1-mx.stm32' ...
INFO: hdimage(sdcard.img): adding partition 'fip' (in MBR) from 'fip.bin' ...
INFO: hdimage(sdcard.img): adding partition 'rootfs' (in MBR) from 'rootfs.ext4' ...
INFO: hdimage(sdcard.img): adding partition '[MBR]' ...
INFO: hdimage(sdcard.img): adding partition '[GPT header]' ...
INFO: hdimage(sdcard.img): adding partition '[GPT array]' ...
INFO: hdimage(sdcard.img): adding partition '[GPT backup]' ...
INFO: hdimage(sdcard.img): writing GPT
INFO: hdimage(sdcard.img): writing protective MBR
INFO: hdimage(sdcard.img): writing MBR
```
Here's the strace output you requested though:
```
sudo strace dd if=output/images/sdcard.img of=/dev/mmcblk0 bs=1M
execve("/usr/bin/dd", ["dd", "if=output/images/sdcard.img", "of=/dev/mmcblk0", "bs=1M"], 0x7fff7af5e7b8 /* 16 vars */) = 0
brk(NULL)                               = 0x55dd97165000
arch_prctl(0x3001 /* ARCH_??? */, 0x7ffc09f2d2d0) = -1 EINVAL (Invalid argument)
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=85073, ...}) = 0
mmap(NULL, 85073, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f4904ef0000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\300A\2\0\0\0\0\0"..., 832) = 832
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
pread64(3, "\4\0\0\0\20\0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0", 32, 848) = 32
pread64(3, "\4\0\0\0\24\0\0\0\3\0\0\0GNU\0\346x\376T\245\322\302\t/\216G\353\v3\20^"..., 68, 880) = 68
fstat(3, {st_mode=S_IFREG|0755, st_size=2029592, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f4904eee000
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
pread64(3, "\4\0\0\0\20\0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0", 32, 848) = 32
pread64(3, "\4\0\0\0\24\0\0\0\3\0\0\0GNU\0\346x\376T\245\322\302\t/\216G\353\v3\20^"..., 68, 880) = 68
mmap(NULL, 2037344, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f4904cfc000
mmap(0x7f4904d1e000, 1540096, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x22000) = 0x7f4904d1e000
mmap(0x7f4904e96000, 319488, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x19a000) = 0x7f4904e96000
mmap(0x7f4904ee4000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1e7000) = 0x7f4904ee4000
mmap(0x7f4904eea000, 13920, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f4904eea000
close(3)                                = 0
arch_prctl(ARCH_SET_FS, 0x7f4904eef580) = 0
mprotect(0x7f4904ee4000, 16384, PROT_READ) = 0
mprotect(0x55dd96036000, 4096, PROT_READ) = 0
mprotect(0x7f4904f32000, 4096, PROT_READ) = 0
munmap(0x7f4904ef0000, 85073)           = 0
rt_sigaction(SIGINT, NULL, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 8) = 0
rt_sigaction(SIGUSR1, {sa_handler=0x55dd960278b0, sa_mask=[INT USR1], sa_flags=SA_RESTORER, sa_restorer=0x7f4904d3f090}, NULL, 8) = 0
rt_sigaction(SIGINT, {sa_handler=0x55dd960278a0, sa_mask=[INT USR1], sa_flags=SA_RESTORER|SA_NODEFER|SA_RESETHAND, sa_restorer=0x7f4904d3f090}, NULL, 8) = 0
brk(NULL)                               = 0x55dd97165000
brk(0x55dd97186000)                     = 0x55dd97186000
openat(AT_FDCWD, "/usr/lib/locale/locale-archive", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=5699248, ...}) = 0
[... 20 lines cut ...]
fstat(3, {st_mode=S_IFREG|0644, st_size=613, ...}) = 0
[... cut ...]
openat(AT_FDCWD, "/usr/share/locale-langpack/en.utf8/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale-langpack/en/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
write(2, ": Operation not permitted", 25: Operation not permitted) = 25
write(2, "\n", 1
)                       = 1
close(2)                                = 0
exit_group(1)                           = ?
+++ exited with 1 +++
```
Unfortunately, this doesn't mean much to me, I'm sure I'm missing something . . . thanks for your help!

tpetazzoni, 2023-10-18:
Which distribution are you running? Do you have some specific security mechanism in place (SELinux, something else) ?

Could you provide the output of:
1. `ls -l output/images`
2. `lsattr output/images`

Could you comment on whether you're doing the Buildroot build as root (or with sudo) or as a regular user?

Overall, this really seems like a specific issue on your system, not something directly Buildroot related.

tpetazzoni, 2024-08-26:
No feedback from the initial bug reporter despite the questions asked, and also the issue really seems like a local system issue, unrelated to the Buildroot integration. Therefore, closing.

closed 2024-08-26 by tpetazzoni