bootlin/buildroot-external-st #45: No support for FIT images building?

aatanas, 2023-10-17:
I could not figure out how to build FIT images using this setup on 157F-DK2, any help would be very much appreciated :)

aatanas, 2023-10-21:
Ok i somehow managed to generate the FIT image manually after the buildroot build. 
Now please do you have any tips about the load adresss? :)

aatanas, 2023-10-21:
Ok I figured out how to do manually after the buildroot build process, but it would be nice for it to be supported natively if its already not ?

kmaincent, 2023-10-23:
There is no generic Makefile way to do fitImage. It is usually done by enabling `BR2_PACKAGE_HOST_UBOOT_TOOLS_FIT_SUPPORT` configuration and building the fitImage in the post-image.sh script.
See these boards as examples:
board/arcturus/aarch64-ucls1012a/post-image.sh
board/aspeed/common/post-image.sh

aatanas, 2023-10-27:
Ok, thank you. Now if i understood correctly, there is also no built-in way in Uboot to load exclusively signed FIT images, you could set it up to require verification but that only applies while loading FIT, nothing is there to prevent loading other types of unsigned images like zImage etc. so source-code modification is required for this type of behavior ?

kmaincent, 2023-10-27:
Yes, but why do you want to lock the other boot mechanism?
If you want to boot only your images you should disable the U-boot prompt access, configure bootcmd to boot your fitImage and sign your fitImage.

aatanas, 2023-10-30:
Ok, thanks, also it seems there is an option to force only FIT images: `CONFIG_LEGACY_IMAGE_FORMAT` :)

kmaincent, 2023-11-03:
I still don't understand why you want to force FIT images but if it is okay for you I can close the issue.

tpetazzoni, 2023-11-03:
> Ok, thank you. Now if i understood correctly, there is also no built-in way in Uboot to load exclusively signed FIT images, you could set it up to require verification but that only applies while [...]

I'm pretty sure there is a U-Boot configuration setting to prevent unsigned images from being loaded. However, this is really a U-Boot question, not a Buildroot/ST question, so I suggest to discuss this on the U-Boot mailing list instead.

closed 2023-11-03 by tpetazzoni