bootlin/buildroot-external-st #46: Linux Frame Buffer (fb0) missing

MfaizR, 2023-10-18:
Hi, 
Can i enable the linux frame buffer (fb) together with the OpenGL? Setting already done but I cant see the device in /dev. please advise.
Thanks

tpetazzoni, 2023-10-18:
Could you clarify what your final goal is? What graphics library do you want to use?

MfaizR, 2023-10-18:
Hi Thomas,
I am trying to put psplash output to the fb0, since that is the psplash default output correct? I understand that it can be redirected to GPU but could not make it work. any advise?

mstaack, 2023-10-18:
for reference: https://github.com/g0hl1n/psplash

buildroot has a build-in boot-image feature afaik. (have seen a menu entry for that)

MfaizR, 2023-10-18:
> for reference: https://github.com/g0hl1n/psplash
> 
> buildroot has a build-in boot-image feature afaik. (have seen a menu entry for that)

Hi @mstaack , 
I never see any of that besides psplash, care to clarify further?
i am looking for the psplash-drm source code, which I believe is for the OpenGL usage.
Thanks

MfaizR, 2023-10-19:
> Hi, Can i enable the linux frame buffer (fb) together with the OpenGL? Setting already done but I cant see the device in /dev. please advise. Thanks

@tpetazzoni @mstaack this has been resolved. I refer to this https://github.com/bootlin/buildroot-external-st/issues/31#issue-1511501266

but still need your help on the uboot save env issue. have update the uboot to [v2022.10-stm32mp-r1.1 but still priblem persists.

Thanks.

closed 2023-10-20 by MfaizR