bootlin/buildroot-external-st #47: Saveenv Issue on v2022.10-stm32mp-r1.1 - Invalid ENV offset in MMC

MfaizR, 2023-10-20:
Hi there,

I am having problem with compiled buildroot-external-st image, where the uboot environment cannot be save with the invalid env offset in mmc. any solutions to this?

thanks

kmaincent, 2023-10-27:
I indeed face the same issue which is also present in v2022.10-stm32mp-r1 version.
It seems related to the U-boot from ST. I will bump the issue to them.

kmaincent, 2023-10-27:
It was due to a missing "u-boot-env" partition.
#[69a6b0d](https://github.com/bootlin/buildroot-external-st/commit/69a6b0d7a8672518a6daa7cb0fdb49a2e1a2bbca) fix the issue.

closed 2023-10-27 by kmaincent

MfaizR, 2023-10-28:
> It was due to a missing "u-boot-env" partition. #[69a6b0d](https://github.com/bootlin/buildroot-external-st/commit/69a6b0d7a8672518a6daa7cb0fdb49a2e1a2bbca) fix the issue.

hi @kmaincent 

I have tested on my board, yes the i can save the environment but my rootfs now its not able to be mounted. Kernel message are as below:

[    2.608832] No filesystem could mount root, tried: 
[    2.608840]  ext3
[    2.613727]  ext2
[    2.615566]  ext4
[    2.617503]  squashfs
[    2.619439]  vfat
[    2.621677]  msdos
[    2.623647]  ntfs
[    2.625585] 
[    2.628947] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(179,4)
[    2.637453] CPU: 1 PID: 1 Comm: swapper/0 Not tainted 6.1.28 #1
[    2.643334] Hardware name: STM32 (Device Tree Support)
[    2.648416]  unwind_backtrace from show_stack+0x10/0x14
[    2.653618]  show_stack from dump_stack_lvl+0x40/0x4c
[    2.658706]  dump_stack_lvl from panic+0xf4/0x300
[    2.663394]  panic from mount_block_root+0x1e0/0x1e8
[    2.668383]  mount_block_root from prepare_namespace+0x150/0x18c
[    2.674374]  prepare_namespace from kernel_init+0x18/0x12c
[    2.679862]  kernel_init from ret_from_fork+0x14/0x2c
[    2.684843] Exception stack(0xde815fb0 to 0xde815ff8)
[    2.689917] 5fa0:                                     00000000 00000000 00000000 00000000
[    2.698122] 5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    2.706225] 5fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[    2.712820] CPU0: stopping
[    2.715559] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 6.1.28 #1
[    2.721422] Hardware name: STM32 (Device Tree Support)
[    2.726576]  unwind_backtrace from show_stack+0x10/0x14
[    2.731747]  show_stack from dump_stack_lvl+0x40/0x4c
[    2.736813]  dump_stack_lvl from do_handle_IPI+0x33c/0x368
[    2.742284]  do_handle_IPI from ipi_handler+0x18/0x20
[    2.747352]  ipi_handler from handle_percpu_devid_irq+0xa4/0x250
[    2.753332]  handle_percpu_devid_irq from generic_handle_domain_irq+0x28/0x38
[    2.760416]  generic_handle_domain_irq from gic_handle_irq+0x74/0x88
[    2.766802]  gic_handle_irq from generic_handle_arch_irq+0x58/0x78
[    2.772982]  generic_handle_arch_irq from __irq_svc+0x88/0xc8
[    2.778651] Exception stack(0xc1201ee8 to 0xc1201f30)
[    2.783708] 1ee0:                   00000000 00000000 00003109 c0119b60 00000000 c1208f40
[    2.791889] 1f00: c1323d80 c1208f40 00000000 00000000 10c5387d 00000000 000cbf0b c1201f38
[    2.800066] 1f20: c0109080 c0109084 60000113 ffffffff
[    2.805118]  __irq_svc from arch_cpu_idle+0x38/0x3c
[    2.809987]  arch_cpu_idle from default_idle_call+0x40/0x12c
[    2.815568]  default_idle_call from cpuidle_idle_call+0x18c/0x1ec
[    2.821648]  cpuidle_idle_call from do_idle+0x94/0xc8
[    2.826713]  do_idle from cpu_startup_entry+0x18/0x1c
[    2.831779]  cpu_startup_entry from rest_init+0xd0/0xd4
[    2.836945]  rest_init from arch_post_acpi_subsys_init+0x0/0x8
[    2.843187] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(179,4) ]---

any advice?

Thanks

MfaizR, 2023-10-30:
Hi @kmaincent 

Tested ok. 
I changed the overlay/boot/extlinux.conf to start root at mmcblk0p5.

Thank you very much for your help.