bootlin/buildroot-external-st #49: stm32mp157f_dk2_demo_defconfig opengl error

mstaack, 2023-12-28:
defconfig fails with opengl error.

system:
```
max@buildroot:~/buildroot$ git status
On branch st/2023.02.2
max@buildroot:~/buildroot$ uname -a
Linux buildroot 5.15.0-91-generic #101-Ubuntu SMP Tue Nov 14 13:29:11 UTC 2023 aarch64 aarch64 aarch64 GNU/Linu
```

```
make BR2_EXTERNAL=../buildroot-external-st st_stm32mp157f_dk2_demo_defconfig
```

last errors:
```
x-gnueabihf/sysroot --sysroot=/home/max/buildroot/output/host/arm-buildroot-linux-gnueabihf/sysroot -w -fPIC  -I. -I/home/max/buildroot/output/build/qt5base-2ffb7ad8a1079a0444b9c72affe3d19b089b60de/mkspecs/devices/linux-buildroot-g++ -o main.o main.cpp
> main.cpp:2:10: fatal error: sybfront.h: No such file or directory
>     2 | #include <sybfront.h>
>       |          ^~~~~~~~~~~~
> compilation terminated.
> gmake[2]: *** [Makefile:183: main.o] Error 1
> gmake[2]: Leaving directory '/home/max/buildroot/output/build/qt5base-2ffb7ad8a1079a0444b9c72affe3d19b089b60de/config.tests/tds'
 => source failed verification.
test config.qtbase_sqldrivers.libraries.tds FAILED
Done running configuration tests.

Configure summary:

Building on: linux-g++ (arm64, CPU features: cx16 neon)
Building for: devices/linux-buildroot-g++ (arm, CPU features: <none>)
Target compiler: gcc 10.3.1
Configuration: cross_compile compile_examples enable_new_dtags largefile shared shared release c++11 c++14 c++17 c++1z concurrent reduce_exports release_tools stl
Build options:
  Mode ................................... release; optimized tools
  Optimize release build for size ........ no
  Building shared libraries .............. yes
  Using C standard ....................... C11
  Using C++ standard ..................... C++17
  Using ccache ........................... no
  Using new DTAGS ........................ yes
  Relocatable ............................ no
  Using precompiled headers .............. no
  Using LTCG ............................. no
  Target compiler supports:
    NEON ................................. no
  Build parts ............................ examples libs
Qt modules and options:
  Qt Concurrent .......................... yes
  Qt D-Bus ............................... no
  Qt D-Bus directly linked to libdbus .... no
  Qt Gui ................................. yes
  Qt Network ............................. yes
  Qt Sql ................................. yes
  Qt Testlib ............................. yes
  Qt Widgets ............................. yes
  Qt Xml ................................. yes
Support enabled for:
  Using pkg-config ....................... yes
  udev ................................... yes
  Using system zlib ...................... yes
  Zstandard support ...................... no
Qt Core:
  DoubleConversion ....................... yes
    Using system DoubleConversion ........ no
  GLib ................................... yes
  iconv .................................. no
  ICU .................................... no
  Built-in copy of the MIME database ..... yes
  Tracing backend ........................ <none>
  Logging backends:
    journald ............................. no
    syslog ............................... no
    slog2 ................................ no
  PCRE2 .................................. yes
    Using system PCRE2 ................... yes
Qt Network:
  getifaddrs() ........................... yes
  IPv6 ifname ............................ yes
  libproxy ............................... no
  Linux AF_NETLINK ....................... yes
  OpenSSL ................................ no
    Qt directly linked to OpenSSL ........ no
  OpenSSL 1.1 ............................ no
  DTLS ................................... no
  OCSP-stapling .......................... no
  SCTP ................................... no
  Use system proxies ..................... yes
  GSSAPI ................................. no
Qt Gui:
  Accessibility .......................... yes
  FreeType ............................... yes
    Using system FreeType ................ yes
  HarfBuzz ............................... no
    Using system HarfBuzz ................ no
  Fontconfig ............................. yes
  Image formats:
    GIF .................................. no
    ICO .................................. yes
    JPEG ................................. no
      Using system libjpeg ............... no
    PNG .................................. no
      Using system libpng ................ no
  Text formats:
    HtmlParser ........................... yes
    CssParser ............................ yes
    OdfWriter ............................ yes
    MarkdownReader ....................... yes
      Using system libmd4c ............... no
    MarkdownWriter ....................... yes
  EGL .................................... no
  OpenVG ................................. no
  OpenGL:
    Desktop OpenGL ....................... no
    OpenGL ES 2.0 ........................ no
    OpenGL ES 3.0 ........................ no
    OpenGL ES 3.1 ........................ no
    OpenGL ES 3.2 ........................ no
  Vulkan ................................. no
  Session Management ..................... yes
Features used by QPA backends:
  evdev .................................. yes
  libinput ............................... no
  INTEGRITY HID .......................... no
  mtdev .................................. no
  tslib .................................. no
  xkbcommon .............................. no
  X11 specific:
    XLib ................................. no
    XCB Xlib ............................. no
    EGL on X11 ........................... no
    xkbcommon-x11 ........................ no
QPA backends:
  DirectFB ............................... no
  EGLFS .................................. no
  LinuxFB ................................ no
  VNC .................................... yes
Qt Sql:
  SQL item models ........................ yes
Qt Widgets:
  GTK+ ................................... no
  Styles ................................. Fusion Windows
Qt PrintSupport:
  CUPS ................................... no
Qt Sql Drivers:
  DB2 (IBM) .............................. no
  InterBase .............................. no
  MySql .................................. no
  OCI (Oracle) ........................... no
  ODBC ................................... no
  PostgreSQL ............................. no
  SQLite2 ................................ no
  SQLite ................................. no
    Using system provided SQLite ......... no
  TDS (Sybase) ........................... no
Qt Testlib:
  Tester for item models ................. yes

Note: Also available for Linux: linux-clang linux-icc

Note: PKG_CONFIG_LIBDIR automatically set to /home/max/buildroot/output/host/arm-buildroot-linux-gnueabihf/sysroot/usr/lib/pkgconfig:/home/max/buildroot/output/host/arm-buildroot-linux-gnueabihf/sysroot/usr/share/pkgconfig:/home/max/buildroot/output/host/arm-buildroot-linux-gnueabihf/sysroot/usr/lib/arm-none-linux-gnueabihf/pkgconfig

Note: PKG_CONFIG_SYSROOT_DIR automatically set to /home/max/buildroot/output/host/arm-buildroot-linux-gnueabihf/sysroot

Note: -optimized-tools is not useful in -release mode.

WARNING: No QPA platform plugin enabled! This will
produce a Qt that cannot run GUI applications.
See "Platform backends" in the output of --help.

ERROR: Feature 'opengles2' was enabled, but the pre-condition '(config.win32 && !features.opengl-dynamic) || (!config.watchos && !features.opengl-desktop && libs.opengl_es2)' failed.

ERROR: Feature 'eglfs' was enabled, but the pre-condition '!config.android && !config.darwin && !config.win32 && !config.wasm && features.egl' failed.

ERROR: Feature 'gbm' was enabled, but the pre-condition 'libs.gbm' failed.

ERROR: The OpenGL functionality tests failed!
You might need to modify the include and library search paths by editing QMAKE_INCDIR_OPENGL[_ES2],
QMAKE_LIBDIR_OPENGL[_ES2] and QMAKE_LIBS_OPENGL[_ES2] in the mkspec for your platform.
make[1]: *** [package/pkg-generic.mk:283: /home/max/buildroot/output/build/qt5base-2ffb7ad8a1079a0444b9c72affe3d19b089b60de/.stamp_configured] Error 3
make: *** [Makefile:82: _all] Error 2
```

tpetazzoni, 2024-01-02:
Thanks for your bug report. You specified in your bug report which version of Buildroot you're using, but not which version of buildroot-external-st. Could you provide this information?

mstaack, 2024-01-03:
just followed the examples steps from the readme:

```
git clone -b st/2023.02.2 https://github.com/bootlin/buildroot-external-st.git
```

tpetazzoni, 2024-01-03:
I just did a test build, with this exact branch for both buildroot and buildroot-external-st, and it builds just fine here. I've pasted the full build log at https://paste.ack.tf/7649e0.

The issue is not sybfront.h: this is a configure check for a database plugin that we don't care about. I have the same message in Qt5's config.log, but my build is fine. The real issues are:

```

ERROR: Feature 'opengles2' was enabled, but the pre-condition '(config.win32 && !features.opengl-dynamic) || (!config.watchos && !features.opengl-desktop && libs.opengl_es2)' failed.

ERROR: Feature 'eglfs' was enabled, but the pre-condition '!config.android && !config.darwin && !config.win32 && !config.wasm && features.egl' failed.

ERROR: Feature 'gbm' was enabled, but the pre-condition 'libs.gbm' failed.

ERROR: The OpenGL functionality tests failed!
```

For me, the configure script reports:

```
  OpenGL:
    Desktop OpenGL ....................... no
    OpenGL ES 2.0 ........................ yes
    OpenGL ES 3.0 ........................ yes
    OpenGL ES 3.1 ........................ yes
    OpenGL ES 3.2 ........................ yes
```

but for you, it reports:

```
  OpenGL:
    Desktop OpenGL ....................... no
    OpenGL ES 2.0 ........................ no
    OpenGL ES 3.0 ........................ no
    OpenGL ES 3.1 ........................ no
    OpenGL ES 3.2 ........................ no
```

Could you double check that you're using the correct Buildroot branch, matching the correct Buildroot-external-st branch? It is important that they are in sync, and being out of sync will typically result in OpenGL not being detected.

mstaack, 2024-01-05:
yeah ok, i am not able to replicate at the moment. thanks

closed 2024-01-05 by mstaack