bootlin/buildroot-external-st #51: Working Config

timsurber, 2024-04-15:
Hello, do you have any advice what has to be done to enable SPDIFRX on STM32MP157F-DK2 (I know, I have to add the hardware receiver module)

My linux devicetree contains:

	spdifrx_pins_mx: spdifrx_mx-0 {
		pins {
			pinmux = <STM32_PINMUX('D', 7, AF9)>; /* SPDIFRX_IN0 */
			bias-disable;
		};
	};

	spdifrx_sleep_pins_mx: spdifrx_sleep_mx-0 {
		pins {
			pinmux = <STM32_PINMUX('D', 7, ANALOG)>; /* SPDIFRX_IN0 */
		};
	};
[...]

	spdif_in: spdif-in {
		#sound-dai-cells = <0>;
		compatible = "linux,spdif-dir";
		status = "okay";

		spdif_in_port: port {
			spdif_in_endpoint: endpoint {
				remote-endpoint = <&spdifrx_endpoint>;
			};
		};
	};

	sound{
		compatible = "audio-graph-card";
		label = "STM32MP15-DK";
		routing = "Playback", "MCLK",
				"Capture", "MCLK",
				"MICL", "Mic Bias";
		dais = <&sai2a_port &sai2b_port &i2s2_port  &spdifrx_port>;
		status = "okay";
	};

[...]

    &spdifrx{
	    pinctrl-names = "default", "sleep";
	    pinctrl-0 = <&spdifrx_pins_mx>;
	    pinctrl-1 = <&spdifrx_sleep_pins_mx>;
	    status = "okay";
    
	    /* USER CODE BEGIN spdifrx */
	    spdifrx_port: port {
		    spdifrx_endpoint: endpoint {
			    remote-endpoint = <&spdif_in_endpoint>;
		    };
	    };
	    /* USER CODE END spdifrx */
    };
Also I compiled CONFIG_SND_SOC_STM32_SPDIFRX into the kernel.

But alsamixer in the capture tab shows no SPDIF recording device.

kmaincent, 2024-04-18:
Hello,

Does the spdifrx driver probe well? 
Do you see any error on the dmesg about it:
`dmesg | grep -i spdi`
`dmesg | grep -i sound`
`dmesg | grep -i snd`

Could you do the same with debug log level enabled if there is nothing from the command above?

timsurber, 2024-04-21:
Hey, thank you for your answer.
Here are the outputs of the commands:    

    # dmesg | grep -i spdi
    [    0.095256] platform 4000d000.audio-controller: Fixed dependency cycle(s) with /spdif-in
    [    2.207794] st,stm32-spdifrx 4000d000.audio-controller: No cache defaults, reading back from HW
    [    4.786594] spdif-dir spdif-in: ASoC: sink widget Capture overwritten

    # dmesg | grep -i sound
    [    0.125976] Advanced Linux Sound Architecture Driver Initialized.
    [    2.472173]   No soundcards found.

    
    # dmesg | grep -i snd
    [    0.276953] usbcore: registered new interface driver snd-usb-audio

    # arecord -l
    **** List of CAPTURE Hardware Devices ****
    card 0: STM32MP15DK [STM32MP15-DK], device 1: 4400b024.audio-controller-cs42l51-hifi cs42l51-hifi-1 [4400b024.audio-controller-cs42l51-hifi cs42l51-hifi-1]
      Subdevices: 1/1
      Subdevice #0: subdevice #0
    card 0: STM32MP15DK [STM32MP15-DK], device 2: 4000b000.audio-controller-i2s-hifi i2s-hifi-2 [4000b000.audio-controller-i2s-hifi i2s-hifi-2]
      Subdevices: 1/1
      Subdevice #0: subdevice #0
    card 0: STM32MP15DK [STM32MP15-DK], device 3: 4000d000.audio-controller-dir-hifi dir-hifi-3 [4000d000.audio-controller-dir-hifi dir-hifi-3]
      Subdevices: 1/1
      Subdevice #0: subdevice #0
    
    # arecord -L
    null
        Discard all samples (playback) or generate zero samples (capture)

kmaincent, 2024-04-23:
```[    4.786594] spdif-dir spdif-in: ASoC: sink widget Capture overwritten```
It seems the sink "capture" is used twice.
Could you try to add this parameter to spdif_in node:
`sound-name-prefix = "spdif0";`

closed 2024-08-26 by tpetazzoni