bootlin/buildroot-external-st #53: rootfs read only

karu2003, 2024-05-02:
You are using rootfs.squashfs. It is read only. Can I change the rootfs type?
i set 
Filesystem images
ext2/3/4 root filesystem 
         ext2/3/4 variant (ext4)  --->
>>>   Executing post-image script /work/buildroot-external-st/board/stmicroelectronics/stm32mp1/post-image-demo.sh
cp: cannot stat '/work/buildroot/output/images/rootfs.squashfs': No such file or directory

I need a big rootfs r/w. I develop in python.

kmaincent, 2024-05-16:
If you change the rootfs format you need to change the name of the rootfs file where it is used.
it is used in the post-image-demo.sh script, in the rauc manifest and in the generation of the sdcard image.
To change it:
```
sed -i s/rootfs.squashfs/rootfs.ext4/g board/stmicroelectronics/stm32mp1/manifest.raucm
sed -i s/rootfs.squashfs/rootfs.ext4/g board/stmicroelectronics/stm32mp1/genimage-demo.cfg
sed -i s/rootfs.squashfs/rootfs.ext4/g board/stmicroelectronics/stm32mp1/post-image-demo.sh
```

tpetazzoni, 2024-08-26:
No more feedback from @karu2003 after our initial reply. And this is not an issue, but a request for doing something else with Buildroot than what is offered by this buildroot-external-st project.

closed 2024-08-26 by tpetazzoni

karu2003, 2024-08-27:
I refused to use the ST platform. It doesn't have many features that I need. In addition, it burned out when I connected it to the USB PD 3.1. It is not suitable for Python. :( ROCK or RPI is easier and has more support.