bootlin/buildroot-external-st #54: Wifi module gets stuck during iperf3 tests

thom24, 2024-05-29:
The wifi module gets stuck during iper3 tests.
The issue mostly appears using --reverse or --bidir options (client is running on the stm32).

The issue was reproduced on STM32MP135F-DK and STM32MP157F-DK2, but some other platforms should be impacted.

Note that this issue was also reproduce using the ST yocto (tag openstlinux-6.1-yocto-mickledore-mp1-v24.03.13).
```
iperf3 -c 192.168.1.105 -t 300 -i 10 --bidir
Connecting to host 192.168.1.105, port 5201
[  5] local 192.168.1.111 port 38108 connected to 192.168.1.105 port 5201
[  7] local 192.168.1.111 port 38114 connected to 192.168.1.105 port 5201
[ ID][Role] Interval           Transfer     Bitrate         Retr  Cwnd
[  5][TX-C]   0.00-10.00  sec  21.2 MBytes  17.8 Mbits/sec    0   1007KBytes
[  7][RX-C]   0.00-10.00  sec  17.9 MBytes  15.0 Mbits/sec
[  5][TX-C]  10.00-20.00  sec  17.2 MBytes  14.4 Mbits/sec    0   1.76MBytes
[  7][RX-C]  10.00-20.00  sec  18.6 MBytes  15.6 Mbits/sec
[  5][TX-C]  20.00-30.00  sec  12.6 MBytes  10.5 Mbits/sec    1   1.41KBytes
[  7][RX-C]  20.00-30.00  sec  14.4 MBytes  12.0 Mbits/sec
[  5][TX-C]  30.00-40.01  sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes
[  7][RX-C]  30.00-40.01  sec  0.00 Bytes  0.00 bits/sec
[  5][TX-C]  40.01-50.01  sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes
[  7][RX-C]  40.01-50.01  sec  0.00 Bytes  0.00 bits/sec
[  5][TX-C]  50.01-60.01  sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes
[  7][RX-C]  50.01-60.01  sec  0.00 Bytes  0.00 bits/sec
^C[  5][TX-C]  60.01-62.56  sec  0.00 Bytes  0.00 bits/sec    0   1.41KBytes
[  7][RX-C]  60.01-62.56  sec  0.00 Bytes  0.00 bits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID][Role] Interval           Transfer     Bitrate         Retr
[  5][TX-C]   0.00-62.56  sec  50.9 MBytes  6.82 Mbits/sec    1   sender
[  5][TX-C]   0.00-62.56  sec  0.00 Bytes  0.00 bits/sec receiver
[  7][RX-C]   0.00-62.56  sec  0.00 Bytes  0.00 bits/sec sender
[  7][RX-C]   0.00-62.56  sec  50.9 MBytes  6.82 Mbits/sec   receiver
iperf3: interrupt - the client has terminated
```

The only way to recover the module is to reset the board.

thom24, 2024-05-29:
From ST point of view, the issue comes from the SDIO Out Of Band interrupt implementation for the WIFI.
A workaround is to pass from Out Of Band interrupt to In Band interrupt.

To do it we need to:
* Apply this patch on the devicetree
```
diff --git a/arch/arm/boot/dts/stm32mp135f-dk.dts b/arch/arm/boot/dts/stm32mp135f-dk.dts                                                                                                                                                                                                  
index bee689f2ee8e..545697f43c7b 100644                                                                                                                                                                                                                                                   
--- a/arch/arm/boot/dts/stm32mp135f-dk.dts                                                                                                                                                                                                                                                
+++ b/arch/arm/boot/dts/stm32mp135f-dk.dts                                                                                                                                                                                                                                                
@@ -494,6 +494,7 @@ &sdmmc2 {                                                                                                                                                                                                                                                             
        pinctrl-1 = <&sdmmc2_b4_od_pins_a &sdmmc2_clk_pins_a>;                                                                                                                                                                                                                            
        pinctrl-2 = <&sdmmc2_b4_sleep_pins_a>;                                                                                                                                                                                                                                            
        non-removable;                                                                                                                                                                                                                                                                    
+       cap-sdio-irq;                                                                                                                                                                                                                                                                     
        st,neg-edge;                                                                                                                                                                                                                                                                      
        bus-width = <4>;                                                                                                                                                                                                                                                                  
        vmmc-supply = <&v3v3_ao>;                                                                                                                                                                                                                                                         
@@ -505,9 +506,6 @@ &sdmmc2 {                                                                                                                                                                                                                                                             
        brcmf: bcrmf@1 {                                                                                                                                                                                                                                                                  
                reg = <1>;                                                                                                                                                                                                                                                                
                compatible = "brcm,bcm4329-fmac";                                                                                                                                                                                                                                         
-               interrupt-parent = <&gpiof>;                                                                                                                                                                                                                                              
-               interrupts = <0 IRQ_TYPE_LEVEL_HIGH>; /* WL_HOST_WAKE */                                                                                                                                                                                                                  
[... cut ...]                                               
                                                                                                                                                                                                                                                                                          
--
```
* Comment the following line in the NVRAM configuration file of the WIFI module (/lib/firmware/brcm/brcmfmac43430-sdio.st,stm32mp135f-dk.txt):
```# muxenab=0x10```

kmaincent, 2024-10-18:
Devicetree has been patched on the ST Linux repository.
The Wifi module configuration has been updated.
Thanks for the report.

closed 2024-10-18 by kmaincent