bootlin/buildroot-external-st #56: U-boot stops: No serial driver found

hilpara, 2024-08-02:
I'm trying to get help this way as I do not get any answers from ST community. I'm using your buildroot release (it  doesn't matter if st/2023.02.10 or st/2024.02.3) and boot stops when u-boot starts as it can't find serial driver. Here is what I wrote to ST community and I hope you could help me to find the reason:

```
....
F/TC:0 0 _regulator_enable:150 vdd
F/TC:0 0 _regulator_enable:188 vdd refcount: 1
F/TC:0 0 _regulator_enable:150 v3v3
F/TC:0 0 _regulator_enable:188 v3v3 refcount: 1
F/TC:0 0 _regulator_enable:150 vddcore
F/TC:0 0 _regulator_enable:188 vddcore refcount: 1
F/TC:0 0 regulator_core_dump:920 Dump Regulators
F/TC:0 0 regulator_core_dump:921 reg      name     use  en      mV      min     max     flags   supply
F/TC:0 0 regulator_core_dump:941 usb33    usb33    0    0       3300    3300    3300    0
F/TC:0 0 regulator_core_dump:941 reg18    reg18    0    1       1800    1800    1800    0       vdd
F/TC:0 0 regulator_core_dump:941 reg11    reg11    0    1       1100    1100    1100    0       vdd
F/TC:0 0 iod_get_voltage:115 sdmmc1_io: get volt
F/TC:0 0 iod_get_state:104 sdmmc1_io: get state
F/TC:0 0 regulator_core_dump:941 sdmmc1_i sdmmc1_i 1    1       3300    0       65535   0x1     vdd
F/TC:0 0 iod_get_voltage:115 sdmmc2_io: get volt
F/TC:0 0 iod_get_state:104 sdmmc2_io: get state
F/TC:0 0 regulator_core_dump:941 sdmmc2_i sdmmc2_i 1    1       3300    0       65535   0x1     vdd
F/TC:0 0 regulator_core_dump:941 vdd      vdd      1    1       3300    3300    3300    0x1
F/TC:0 0 regulator_core_dump:941 v3v3     v3v3     1    1       3300    3300    3300    0x1
F/TC:0 0 regulator_core_dump:941 vddcore  vddcore  1    1       1350    1350    1350    0x1
D/TC:0 0 call_initcalls:40 level 6 scmi_regulator_consumer_init()
F/TC:0 0 get_supply_phandle:114 voltd: supplied by 7
D/TC:0 0 scmi_regulator_consumer_init:322 scmi voltd shares reg11 (node reg11) on domain ID 0
F/TC:0 0 get_supply_phandle:114 voltd: supplied by 8
D/TC:0 0 scmi_regulator_consumer_init:322 scmi voltd shares reg18 (node reg18) on domain ID 1
F/TC:0 0 get_supply_phandle:114 voltd: supplied by 9
D/TC:0 0 scmi_regulator_consumer_init:322 scmi voltd shares usb33 (node usb33) on domain ID 2
D/TC:0 0 call_initcalls:40 level 7 stm32_rng_release()
D/TC:0 0 call_initcalls:40 level 7 regulator_core_cleanup()
F/TC:0 0 regulator_core_cleanup:1008
D/TC:0 0 regulator_core_cleanup:1012 disable usb33
D/TC:0 0 regulator_core_cleanup:1012 disable reg18
D/TC:0 0 regulator_core_cleanup:1012 disable reg11
D/TC:0 0 call_initcalls:40 level 7 release_probe_lists()
D/TC:0 0 call_finalcalls:59 level 1 release_external_dt()
I/TC: Primary CPU switching to normal world boot

<debug_uart>
serial_stm32 serial@40010000: pinctrl_select_state_full: uclass_get_device_by_phandle_id: err=-19
serial_stm32 serial@40010000: pinctrl_select_state_full: uclass_get_device_by_phandle_id: err=-19
serial_stm32 serial@40010000: pinctrl_select_state_full: uclass_get_device_by_phandle_id: err=-19
serial_stm32 serial@40010000: pinctrl_select_state_full: uclass_get_device_by_phandle_id: err=-19
No serial driver found
resetting ...
System reset not supported on this platform
### ERROR ### Please RESET the board ###
```

I have tried to debug this by adding print statements to serial-uclass.c.

In a function serial_check_stdout the chosen console is found serial0:115200n8, node 7780. Later the device uclass ID is UCLASS_SERIAL, but then the device_probe returns -ENODEV = No such device.

I did dump the tree of all devices

```
Class     Index  Probed  Driver                Name
-----------------------------------------------------------
 root          0  [ + ]   root_driver           root_driver
 simple_bus    0  [   ]   simple_bus            |-- soc
 clk           0  [   ]   fixed_clock           |-- clk-mco1
 serial        0  [   ]   serial_stm32          `-- serial@40010000
```
So the serial is found. 

uclass

```
uclass 95: serial
0     serial@40010000 @ c02803fc, seq 0
```
Driver

`serial_stm32              095 serial               serial@40010000`

I went down the rabbit hole and managed to find, that the problem is clock, but I still don't know what is wrong:

uclass_find_device_by_ofnode (uclass.c) tries to find device node for [rcc@50000000](mailto:rcc@50000000). uclass_get will find it with the key 29
```
Driver                    uid uclass               Devices
----------------------------------------------------------
stm32mp13_clk             029 clk                  <none>
```

and uclass_foreach_dev(dev, uc) do not anything as us->(I don't know which one) is NULL or 0 and then the uclass_find_device_by_ofnode will return -ENODEV.

I have also included my defconfig and device tree files. I hope you could point me to a right direction why the boot stops.

[stm32mp135_defconfig.txt](https://github.com/user-attachments/files/16469961/stm32mp135_defconfig.txt)
[stm32mp135-optee.dts.txt](https://github.com/user-attachments/files/16469962/stm32mp135-optee.dts.txt)
[stm32mp135-tfa.dts.txt](https://github.com/user-attachments/files/16469963/stm32mp135-tfa.dts.txt)
[stm32mp135-uboot-u-boot.dts.txt](https://github.com/user-attachments/files/16469964/stm32mp135-uboot-u-boot.dts.txt)
[stm32mp135-uboot.dts.txt](https://github.com/user-attachments/files/16469965/stm32mp135-uboot.dts.txt)

tpetazzoni, 2024-08-02:
First of all, are you using an ST evaluation board, or a custom board design ?

In all cases, I would recommend to first make sure that things are working as expected on a ST evaluation board. Could you try, on the STM32MP135 evaluation board from ST:
- Our prebuilt image https://bootlin.com/pub/buildroot-st/openstlinux-6.1-buildroot-2023.02.10-mp1-v23.11.15/sdcard-st_stm32mp135f_dk.img.gz ?
- Then try building our st_stm32mp135f_dk_defconfig configuration, and see if it also works for you?

Which version of STM32CubeMX have you used to generate your Device Tree file? There is some level of dependency between the version of STM32CubeMX and the version of the ST BSP components (TF-A, U-Boot, Linux) that are used, so you can use whatever version of STM32CubeMX with whatever version of the ST BSP components.

Also the pinctrl_select_state_full messages are a bit worrying, you are getting an ENODEV from the pinctrl subsystem, this also doesn't look good.

hilpara, 2024-08-02:
I'm using development board from MyIR (MYD-YF13X https://www.myirtech.com/list.asp?id=727). I do not have access to evaluation board from ST.

For the device trees, I used the mix of bootlin device trees and the ones from the MyIR. If there would be a PMIC I could directly use the bootlin files.
All I have configured or at least try to configure is UART4 for debugging and mmc1 (SD-card) and mmc2 (EMMC). I'm just trying to get it boot (from SD-card) first and then add the rest later.

The biggest difference to bootlin device trees are missing PMIC (fixed regulators and maybe DDR, as I took that configuration from MyIR files) and therefore in op-tee:
```
cpu0-opp-table {
                opp-1000000000 {
                        opp-hz = /bits/ 64 <1000000000>;
                        opp-microvolt = <1350000>;
                        opp-supported-hw = <0x2>;
                        st,opp-default;
                };
		/delete-node/ opp-650000000;
    };
```

As you said, ENODEV comes from pinctrl, but UART4 clocks and the pins are not controlled by the OP-TEE so it can't be that. UART4 is also working from u-boot, otherwise I would not be able to see the error messages. So I can't really understand why it can't find serial driver. Also it stops when it probes rcc part of device tree.

I did try to generate files from the CubeMX, but they are missing so much configuration that it doesn't make any sense.

tpetazzoni, 2024-08-26:
Unfortunately @hilpara, we are unable to help you. Supporting the MyIR board is outside the scope of this buildroot-external-st project. Bootlin can provide commercial support to implement Linux BSP for customer STM32MP platforms, do not hesitate to contact us.

closed 2024-08-26 by tpetazzoni