bootlin/buildroot-external-st #59: "st_stm32mp135f_dk_defconfig" can not build

RichSj, 2024-09-27:
I use "st_stm32mp157f_dk2_defconfig" build image OK, but when I use "st_stm32mp135f_dk_defconfig", I got error message "
>>>   Executing post-image script /home/rich1/buildroot-external-st/board/stmicroelectronics/common/post-image.sh
Can not find ATF binary tf-a-stm32mp135f-dk.stm32"

kmaincent, 2024-09-30:
Hello, is your build based on the last branch st/2024.02.3 release?
TF-A has not been built as it can't find the image.
Could you check if a TF-A image is present:
$ ls buildroot/output/images

RichSj, 2024-10-01:
If I build only one config file, it is work.

1. I use st_stm32mp157f_dk2_defconfig build, then use st_stm32mp135f_dk_defconfig, I got error.
2. I use st_stm32mp135f_dk_defconfig build, then use st_stm32mp157f_dk2_defconfig, I got error.
3, Delete package, use git get new package, build again, it work fine.

I guess after build one set, some files keep there, 
This happen in many cases, develop and user got different result. 

Regards,

Richard

closed 2024-10-01 by RichSj

kmaincent, 2024-10-02:
All the building files are saved in the `output` directory by default.
Indeed you can't use two different defconfig for one output directory.
If you want to build several configurations use the O options
```
make BR2_EXTERNAL=../buildroot-external-st O=output1 st_stm32mp157f_dk2_defconfig
make O=output1
make BR2_EXTERNAL=../buildroot-external-st O=output2 st_stm32mp135f_dk_defconfig
make O=output2
```

RichSj, 2024-10-03:
Thank yoy!