bootlin/buildroot-external-st #60: Suspend-to-RAM broken: Instead of resuming properly, the device reboots

DrTobe, 2024-10-10:
Instead of resuming where the system left off before suspending, the device does a full reboot. This has been tested with the pre-built images for the DK and the DK2 board.

Steps to reproduce:
1. Download the image, flash the SD card, boot the system
2. `echo deep > /sys/power/mem_sleep`
3. `echo +20 > /sys/class/rtc/rtc0/wakealarm`
4. `echo mem > /sys/power/state`
5. Wait 20 seconds, the system reboots instead of resuming

I initally observed this on an STM32MP157C-DK2 board (157C, not 157F which I do not have available). To verify that behaviour on the exact reference hardware, I tested this on the STM32MP135F-DK, too. It shows the same behaviour so I suppose this may be true for all setups.

If you need any more information or if I should do some additional testing, please let me know.

kmaincent, 2024-10-18:
Hello, thanks for reporting the issue.
The problem comes from using STM32MP_USB_PROGRAMMER build option in TF-A.
Indeed using this option disabled the power features to avoid oversized image.
We will need a dedicated TF-A image for flashing also on stm32mp1 then.
I will update the branch.

DrTobe, 2024-10-18:
Incredible, just removing `STM32MP_USB_PROGRAMMER=1` from the additional TF-A build variables fixes the suspend-to-RAM mode.

Thank you so so so so much!

closed 2024-10-18 by DrTobe