bootlin/buildroot-external-st #61: make linux-menuconfig fails

ratcro, 2024-11-03:
I just set up a new Ubuntu 24.04.1 LTS machine to develop for the STM32MP157F-DK2 board (using [bootlin instructions](https://bootlin.com/blog/updated-buildroot-support-for-stm32mp1-platforms/)).  I have been able to successfully build an SD card image, so the basic build environment appears to be complete.  Now, I would like to configure the kernel to experiment with using this board as a USB mass storage gadget (with the [DWC2 driver](https://wiki.stmicroelectronics.cn/stm32mpu/index.php?title=USB_overview&oldid=75157)).  To do this, it looks like I need to run `make linux-menuconfig`.

This throws an error, where it looks like maybe an environment variable isn't set (it is trying to resolve an absolute path starting with `/board/stmicroelectronics/stm32mp1` that doesn't exist).

`$ make menuconfig`
`for f in  /home/user/buildroot/output/build/linux-custom/arch/arm/configs/fragment-01-multiv7_cleanup.config /home/user/buildroot/output/build/linux-custom/arch/arm/configs/fragment-02-multiv7_addons.config /board/stmicroelectronics/stm32mp1/linux-disable-etnaviv.config; do if [ ! -f "${f}" ]; then printf "Kconfig file or fragment '%s' for '%s' does not exist\n" "${f}" "linux"; exit 1; fi; done
Kconfig file or fragment '/board/stmicroelectronics/stm32mp1/linux-disable-etnaviv.config' for 'linux' does not exist
make[1]: *** [linux/linux.mk:597: /board/stmicroelectronics/stm32mp1/linux-disable-etnaviv.config] Error 1
make: *** [Makefile:84: _all] Error 2`

This error appears to be generated in `package/pkg-kconfig.mk`, line 141.  It seems to be looking for a file that does exist but with the `/home/user/buildroot-external-st` prefix missing.

ratcro, 2024-11-05:
Short of rebuilding the VM from scratch and starting over, my `.bash_history` paints a fairly good picture of what I did.  I have edited out mis-typed commands and anything not relevant (like `df -h` and `ls`).  I also had to increase the amount of disk space since the Ubuntu installer left part of the virtual disk unallocated and I ran out of space during the build.

sudo apt update
sudo apt upgrade
git clone -b st/2022.02 https://github.com/bootlin/buildroot.git
git clone -b st/2022.02 https://github.com/bootlin/buildroot-external-st
cd buildroot
make
sudo apt install make build-essential
make BR2_EXTERNAL=../buildroot-external-st st_stm32mp157f_dk2_demo_defconfig
make
sudo apt install unzip
make

It took me a few tries to build everything as I discovered which packages had not been installed as part of the default Ubuntu 24.04.1 server distribution.  At this point, though, everything was compiling and this was when I tried doing the `make linux-menuconfig` that failed.

tpetazzoni, 2024-11-05:
Could you use the latest/current version instead of 2022.02 ? Please check the documentation at https://github.com/bootlin/buildroot-external-st. Thanks!

ratcro, 2024-11-05:
I'll be on travel for the rest of the week so I can't dig into this much further right now, but I tried to switch the existing checkouts I had to the `st/2024.02.3` branch, and now whenever I invoke `make` I get this error (retyped, so pardon any typos):

make clean

package/optee-client/optee-client.mk:61: *** OPTEE_CLIENT_SITE (https://github.com/OP-TEE/optee_client/archive/) cannot have a trailing slash.  Stop.
make: *** [Makefile:82: _all] Error 2

I admittedly did this without taking the time to read all of the instructions you linked to so I might not have set it up right.  This weekend, I'll wipe everything, check out the correct (latest) branch cleanly, and try again.

ratcro, 2024-11-16:
Just to follow up, I pulled down the st/2024.02.03 branches and everything seems to build cleanly.  The linux-menuconfig target comes up as expected and I have been able to add the kernel features I needed.  Thank you for the assist.

closed 2024-11-16 by ratcro