bootlin/buildroot-external-st #62: Access serial/uart ports on STM32MP157-Dev DK2 board

SPJAU, 2025-02-10:
Hi, I am trying to connect simple serial device to the dev board. How do I access and print to consol/terminal output from the serial device.

mstaack, 2025-02-10:
you can simply use the onboard stlink device connected via usb. it enumerates as a serial device on you computer

mstaack, 2025-02-10:
rereading your question:

for normal uart to work it needs to be enabled in the devicetree of course.

check this: https://wiki.st.com/stm32mpu/wiki/Serial_TTY_device_tree_configuration#DT_configuration_-28Board_level-29

SPJAU, 2025-02-10:
According to manual I can modify or change device tree using CubeMx, however there are no clear explanation how to do it. It could be handy to have this option to enable low level devices without compiling Linux core. I believe easiest way going to be edit stm32mp157c-dk2.dts file. How do I make sure that this file has been selected during compilation? because in my case I cannot access serial port except ttySTM. Thanks in advance.

mstaack, 2025-02-11:
editing the device tree does not require compiling the linux core.

highly recommend the wonderful bootlin article here:

https://bootlin.com/blog/building-a-linux-system-for-the-stm32mp1-basic-system/

this shows how to work witht he stm32mp1 dev boards and enabling custom io (i2c in this case) in the device-tree

SPJAU, 2025-02-21:
Ok I understand. Thanks for the info.

closed 2025-05-13 by kmaincent